pioarduino/platform-espressif32 #171: Problem when using pioarduino with ESP32C6

majuss, 2025-05-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-05-08:
No support for alpha version

closed 2025-05-08 by Jason2866

majuss, 2025-05-08:
@Jason2866 exactly the same error with release version 54.03.20

TD-er, 2025-05-08:
Does ESPHome already support the latest ESP-IDF5.x / Arduino 3.x ?