pioarduino/platform-espressif32 #207: PVS Studio with windows_amd64

Laxilef, 2025-06-30:
Hi all,

Related: https://github.com/platformio/platformio-core/issues/5194
Since version 7.18.59866, the package ``platformio/tool-pvs-studio`` is not compatible with ``windows_amd64``:
```
UnknownPackageError: Could not find the package with 'platformio/tool-pvs-studio @ ^7.18.59866' requirements for your system 'windows_amd64'
```

And version 7.18.59865 does not work with the latest version of pio:
```
Checking s3_mini > pvs-studio (platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip; board: lolin_s3_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------
In file included from D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/machine/_default_types.h:8,
                 from D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/stdint.h:12,  
                 from D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/lib/gcc/riscv32-esp-elf/14.2.0/include/stdint.h:9,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:24,
                 from D:\Projects\pvstest\src\main.cpp:4:
D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/sys/features.h:590:9: warning: "_POSIX_READER_WRITER_LOCKS" redefined
  590 | #define _POSIX_READER_WRITER_LOCKS              200112L
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~
<command-line>: note: this is the location of the previous definition
In file included from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:194:
D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/WCharacter.h:24:9: warning: "isascii" redefined
   24 | #define isascii(__c) ((unsigned)(__c) <= 0177)
      |         ^~~~~~~
In file included from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/WCharacter.h:23:
D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/ctype.h:99:9: note: this is the location of the previous definition
   99 | #define isascii(__c)    ((unsigned)(__c)<=0177)
      |         ^~~~~~~
D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/WCharacter.h:25:9: warning: "toascii" redefined
   25 | #define toascii(__c) ((__c) & 0177)
      |         ^~~~~~~
D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/ctype.h:100:9: note: this is the location of the previous definition
  100 | #define toascii(__c)    ((__c)&0177)
      |         ^~~~~~~
In file included from D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/errno.h:42,   
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/newlib/platform_include/errno.h:10,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/lwip/port/esp32xx/include/arch/cc.h:12,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/lwip/lwip/src/include/lwip/arch.h:48,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/lwip/lwip/src/include/lwip/debug.h:40,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/lwip/lwip/src/include/lwip/opt.h:52,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/lwip/lwip/src/include/lwip/ip_addr.h:40,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/IPAddress.h:25,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:199:
D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/picolibc/include/sys/errno.h:67:9: warning: "_REENT_ERRNO" redefined
   67 | #define _REENT_ERRNO(r) (errno)
      |         ^~~~~~~~~~~~
In file included from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/newlib/platform_include/sys/reent.h:13,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/xtensa/include/xtensa_config.h:140,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/freertos/config/xtensa/include/freertos/FreeRTOSConfig_arch.h:14,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/freertos/config/include/freertos/FreeRTOSConfig.h:62,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32-libs/esp32s3/include/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:63,
                 from D:/Projects/pvstest/.pio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33:
D:/Projects/pvstest/.pio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/sys/reent.h:818:9: note: this is the location of the previous definition
  818 | #define _REENT_ERRNO(_ptr) (_tls_errno)
      |         ^~~~~~~~~~~~

Error: pvs-studio failed to perform check! Please examine tool output in verbose mode.
========================================= [FAILED] Took 7.71 seconds ========================================= 

Environment    Tool        Status    Duration
-------------  ----------  --------  ------------
s3_mini        pvs-studio  FAILED    00:00:07.705
==================================== 1 failed, 0 succeeded in 00:00:07.705 ====================================
```

Visual Studio Code: 1.101.2
Electron: 35.5.1
ElectronBuildId: 11727614
Chromium: 134.0.6998.205
Node.js: 22.15.1
V8: 13.4.114.21-electron.0
ОС: Windows_NT x64 10.0.22631
PlatformIO Core: 6.1.18
PlatformIO Home: 3.4.4
[Arduino Release v3.2.0 based on ESP-IDF v5.4.1](https://github.com/pioarduino/platform-espressif32/releases/tag/54.03.20)

Can this be fixed?

Laxilef, 2025-06-30:
Why x86? I'm using x64 :)
However, in VS Code I used ``PlatformIO IDE`` extension with ``platform = https://github.com/pioarduino/platform-espressif32/releases/download/54.03.20/platform-espressif32.zip``, not ``pioarduino IDE``. I will test with ``pioarduino IDE`` now and give feedback.

Jason2866, 2025-07-01:
The issue is:
```
UnknownPackageError: Could not find the package with 'platformio/tool-pvs-studio @ ^7.18.59866' requirements for your system 'windows_amd64'
```
the package i made for PVS-Tool is wrong/defect.

TD-er, 2025-07-01:
PVS Studio was also giving issues when used on Windows when using WSL.
I only got it to work using PlatformIO with everything building on Windows.
Which is slowwwwww...... (> 45 mins for a full analysis of my ESPEasy project)

I have had contact with the people from PVS Studio about this and they helped me to run it via command-line which was much-much faster on WSL, but sadly the output was completely unfiltered so very hard to use.

N.B. you can get a free license from them for use with OpenSource software.
Has to be renewed yearly and part of their agreement to give out a free license is that you make a mention of PVS Studio whenever you fixed some issues reported by their tool.

Have to say I am still quite impressed by what it can find.

Jason2866, 2025-07-01:
There is no `platform_packages` needed. Either it works now with the updated package or not.
As said before, can't get any PVS-TOOL working so i have no idea whats wrong (or not).

Laxilef, 2025-07-01:
Well, without ``platform_packages`` I see an error:
```
Tool Manager: Installing platformio/tool-pvs-studio @ ^7.18.59866
UnknownPackageError: Could not find the package with 'platformio/tool-pvs-studio @ ^7.18.59866' requirements for your system 'windows_amd64'
```

Laxilef, 2025-07-01:
> The error you see, you have an outdated platform. Delete folder `.platformio/platforms`

Of course I tried deleting ``C:\Users\Yurii\.platformio`` and ``.pio`` in the project directory (because I use ``core_dir = .pio``)
I don't know why the extension tries to download ``platformio/tool-pvs-studio@^7.18.59866``, I don't use it in ``platformio.ini``

Here is the full console log after deleting the directories described above and starting VS Code: [log.txt](https://github.com/user-attachments/files/21008660/log.txt)

Jason2866, 2025-07-02:
I will probably remove the not working support for PVS-TOOL It is a mess since it never works for me. Either with orig. Platformio Platform nor with pioarduino.

Laxilef, 2025-07-06:
Hi @Jason2866,
I sent a letter to PVS support with a link to this isuue. Maybe they will be interested in fixing the integration of their tool into pioarduino 🔥

Laxilef, 2025-09-24:
> Try branch `mv_py_setup` Close VSC and remove (or rename) hidden folder `.platformio` Start VSC and open an existing pioarduino project (Open Folder). The installer is now called after the penv is installed and called with venv Python.

Done. On run Check:

```
FileNotFoundError: Traceback (most recent call last):
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ~~~^^
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ~~~~~~~~~~~~~~^^^^^
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\platformio\check\cli.py", line 156, in cli
    rc = ct.check(
        on_defect_callback=(
    ...<3 lines>...
        )
    )
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\platformio\check\tools\pvsstudio.py", line 252, in check
    result = self.execute_check_cmd(cmd)
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\platformio\check\tools\base.py", line 192, in execute_check_cmd
    result = proc.exec_command(
        cmd,
        stdout=proc.LineBufferedAsyncPipe(self.on_tool_output),
        stderr=proc.LineBufferedAsyncPipe(self.on_tool_output),
    )
  File "C:\Users\Yurii\.platformio\penv\Lib\site-packages\platformio\proc.py", line 114, in exec_command
    with subprocess.Popen(*args, **kwargs) as p:
         ~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python313\Lib\subprocess.py", line 1039, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
    ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
                        pass_fds, cwd, env,
                        ^^^^^^^^^^^^^^^^^^^
    ...<5 lines>...
                        gid, gids, uid, umask,
                        ^^^^^^^^^^^^^^^^^^^^^^
                        start_new_session, process_group)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Program Files\Python313\Lib\subprocess.py", line 1554, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
                       ~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
                             # no special security
                             ^^^^^^^^^^^^^^^^^^^^^
    ...<4 lines>...
                             cwd,
                             ^^^^
                             startupinfo)
                             ^^^^^^^^^^^^
FileNotFoundError: [WinError 2]
```

``.platformio/tools/tool-pvs-studio`` dir is created, but empty. In ``.platformio/packages/tool-pvs-studio`` there are only ``.piopm``, ``package.json``, ``tools.json`` files