pioarduino/platform-espressif32 #230: `analogWriteResolution` not working

vladtepesch, 2025-07-24:
### PROBLEM DESCRIPTION

I try to setup a 10bit pwm with `analogWriteResolution(LED_R, 10);` but I get the following output:

>    [esp32-hal-ledc.c:793] analogWriteResolution(): analogWrite resolution cant be set due to selected frequency! Try to adjust frequency first

then I tried to explicitely set the frequency `analogWriteFrequency(LED_R,1000);`

then I get an additional message:

>    [esp32-hal-ledc.c:785] analogWriteFrequency(): analogWrite frequency cant be set due to selected resolution! Try to adjust resolution first

```
void setup() {
  ledcSetClockSource(LEDC_USE_APB_CLK);
  analogWriteFrequency(LED_R,1000);
  analogWriteResolution(LED_R, 10);
}
```

I really do not understand the problem. 10bit @ 1khz requires a timerfrequency of 1Mhz which should be achievable without problems.

my platormio.ini

```
[env:esp32]

platform =  https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = upesy_wroom
board_build.mcu = esp32

board_build.f_cpu = 240000000L

framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
#build_type = debug
#upload_port = COM6
board_build.filesystem = littlefs

build_unflags = -std=gnu++11
                -std=c++11

build_flags =
    -std=c++17
    -std=gnu++17

``` 

### TO REPRODUCE
see code above

### EXPECTED BEHAVIOUR
correct pwm setup

### SCREENSHOTS
n.a.

### ADDITIONAL CONTEXT

[I posted this Problem on SO](https://stackoverflow.com/questions/79711915/can-not-set-frequency-or-resolution-for-output-pin?noredirect=1#comment140613703_79711915)

A sugestion was to set resolution and frequency by directly using `ledcAttach`. this works as expecteds - so we came to the conclusion that most probably this is a library bug.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-07-24:
not a pioarduino issue. It is an Arduino core question / issue. Please open a issue in espressif GitHub repo

vladtepesch, 2025-07-27:
I am not sure that the correct target repo (currenty do not fully understand the pgraph that does depend on what and that is forked from what) is. can you forward the issue to the correct one and provide addtional necessary informations (like the version this repo is based on)? I will then try to follow up on this issue.

Jason2866, 2025-07-27:
The correct repo for this is https://github.com/espressif/arduino-esp32

TD-er, 2025-07-27:
Possible related issue
https://github.com/espressif/arduino-esp32/issues/9336