pioarduino/platform-espressif32 #235: Problems on macOS with 3.3.0...

UnexpectedMaker, 2025-07-26:
I am seeing the same thing if I use the stable release, and if I use this:
`platform = https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30/platform-espressif32.zip ; 3.3.0`

I get lots of these when installing...
`idf_tools.py installation failed`
And when I compile I get this:
```
rocessing rmp_release (platform: https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30/platform-espressif32.zip; board: um_tinys2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------
idf_tools.py installation failed
idf_tools.py installation failed
Verbose mode can be enabled via `-v, --verbose` option
idf_tools.py installation failed
error: /Users/seon/.platformio/packages/tool-esptoolpy does not appear to be a Python project, as neither `pyproject.toml` nor `setup.py` are present in the directory
Warning: Failed to install esptool: Command '['/Users/seon/.platformio/penv/bin/uv', 'pip', 'install', '--quiet', '--python=/Users/seon/.platformio/penv/bin/python', '-e', '/Users/seon/.platformio/packages/tool-esptoolpy']' returned non-zero exit status 2.
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/um_tinys2.html
PLATFORM: Espressif 32 (55.3.30) > Unexpected Maker TinyS2
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - contrib-piohome @ 3.4.4 
 - framework-arduinoespressif32 @ 3.3.0 
 - framework-arduinoespressif32-libs @ 5.5.0+sha.b66b5448e0 
 - tool-esptoolpy @ 5.0.1 
 - tool-mklittlefs @ 3.2.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 58 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 3.3.0
|-- OneButton @ 2.6.1
|-- Adafruit BusIO @ 1.17.2
|-- Adafruit GFX Library @ 1.12.1
|-- Adafruit ILI9341 @ 1.6.2
|-- Adafruit BME280 Library @ 2.3.0
|-- PID @ 1.2.1
|-- PubSubClient @ 2.8.0
|-- AsyncTCP @ 3.4.5
|-- ESPAsyncWebServer @ 3.7.10
|-- SPI @ 3.3.0
|-- WiFi @ 3.3.0
|-- ESPmDNS @ 3.3.0
|-- HTTPClient @ 3.3.0
|-- FS @ 3.3.0
|-- SD @ 3.3.0
|-- FXL6408
|-- Adafruit Unified Sensor @ 1.1.15
|-- DNSServer @ 3.3.0
Building in release mode
Building .pio/build/rmp_release/bootloader.bin
*** [.pio/build/rmp_release/partitions.bin] Source `default_4MB.csv' not found, needed by target `.pio/build/rmp_release/partitions.bin'.
sh: esptool: command not found
*** [.pio/build/rmp_release/bootloader.bin] Error 127
========================================== [FAILED] Took 3.48 seconds ==========================================

Environment    Status    Duration
-------------  --------  ------------
rmp_release    FAILED    00:00:03.483
===================================== 1 failed, 0 succeeded in 00:00:03.483 =====================================
```

I have deleted my .platformio folder twice now, restarted my Mac, and re-started VSC several time.

I've made sure my PIO core is latest as well.

I am able to use 3.2.1 and earlier correctly, but 3.3.0 is borked for me.

UnexpectedMaker, 2025-07-26:
I just get empty install folders with no `bin` folders.  Example:

[image]

As you can see, this is all a clean download this morning:

[image]

UnexpectedMaker, 2025-07-26:
If I do a PIO clean I get this:
```
 *  Executing task in folder Reflow_Master_Pro: platformio run --target clean --environment rmp_release 

Processing rmp_release (platform: https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30/platform-espressif32.zip; board: um_tinys2; framework: arduino)
-----------------------------------------------------------------------------------------------------------------
idf_tools.py installation failed
Verbose mode can be enabled via `-v, --verbose` option
idf_tools.py installation failed
Removing .pio/build/rmp_release
Done cleaning
========================================== [SUCCESS] Took 0.34 seconds ==========================================

Environment    Status    Duration
-------------  --------  ------------
rmp_release    SUCCESS   00:00:00.344
========================================== 1 succeeded in 00:00:00.344 ==========================================
 *  Terminal will be reused by tasks, press any key to close it. 
```

So something is broken with the python scripts for installing toolchain stuff.

Jason2866, 2025-07-26:
idf_tools.py installation failed. pioarduino uses the very minimal modified espressif idf installer. Probably certs are missing. See https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html

UnexpectedMaker, 2025-07-26:
> idf_tools.py installation failed. pioarduino uses the very minimal modified espressif idf installer. Probably certs are missing. See https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/linux-macos-setup.html

Hey Jason, thanks for the quick reply.... I've been working on previous PIO versions daily on many projects without issue until I installed 3.0.0 yesterday.

So are you saying I have to have a full esp-idf installed on my machine to use this? That is what a certificates install requires.

And I am not getting any certificate/ssl errors like suggested in the idc install docs.

Jason2866, 2025-07-26:
No, just extract the certs from the tool and import in the local cert store.

The AI told to do like this

Here’s how to import a certificate locally on a Mac:
Import Certificate via Keychain Access
1. Open Keychain Access
	•	Open Keychain Access app via Spotlight (Cmd+Space and search for “Keychain Access”)
	•	Alternatively, find it under Applications > Utilities > Keychain Access
2. Select Keychain
Choose the appropriate keychain:
	•	Login: For user-specific certificates
	•	System: For system-wide certificates
3. Import Certificate
Several methods available:
Method 1: Drag & Drop
	•	Drag the certificate file directly into the Keychain Access window
Method 2: Via Menu
	•	Go to File > Import Items…
	•	Select your certificate file (.p12, .pfx, .crt, .pem)
	•	Click Open
Method 3: Double-click
	•	Double-click directly on the certificate file - this starts the import automatically
4. Enter Passwords
During import, you’ll be prompted for passwords:
	•	First password: Password for the certificate file (if applicable)
	•	Second password: Your Mac admin account password for confirmation
Supported Certificate Formats
	•	.p12/.pfx: Personal Information Exchange files
	•	.crt: Certificate files
	•	.pem: Privacy Enhanced Mail format
Verify Certificate
After import:
	•	Select My Certificates category
	•	The imported certificate should appear in the list
Adjust Trust Settings
If needed, you can modify the certificate’s trust settings:
	•	Double-click the imported certificate
	•	Expand the Trust section
	•	Set desired trust levels
The imported certificate is now available system-wide for applications like Safari, Apple Mail, and other programs that use the macOS keychain.

Jason2866, 2025-07-26:
Forgot to say store the marked part in a file with the extension *.pem

UnexpectedMaker, 2025-07-26:
I think you are sending me on a while goose chase - I don't beleive this is a certificates issue.

My IDF 5.4.x version (that lives outside of PIO) builds and deploys correctly to devices. I just tried it, so not a certificates issue.

There is something else going on here that we need to get to the bottom of.

Jason2866, 2025-07-27:
It is a certificate issue, had to update the installer because ninja package was updated and  zip package can only correctly installed with a very actual installer version.

closed 2025-07-27 by Jason2866

UnexpectedMaker, 2025-07-27:
So what does that mean? Do I need to wait for a new release to get this fix?

Jason2866, 2025-07-27:
No, do what i wrote. Install the missing certificates. That's it.

UnexpectedMaker, 2025-07-27:
I just grabbed latest openSSL certs, installed and verified they are installed globally and are visible via all python versions...

I then deleted my entire `~/.platformio` folder and loaded my project and told it to use 3.2.1 and it started downloading, unpacking and installing...

```
Resolving rmp_release dependencies...
Platform Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21/platform-espressif32.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: espressif32@54.3.21 has been installed!
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.2.1/esp32-3.2.1.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.2.1 has been installed!
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.4/esp32-arduino-libs-idf-release_v5.4-858a988d-v1.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32-libs@5.4.0+sha.858a988d6e has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v5.0.0-dev1/esptool.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@5.0.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp_install/releases/download/v5.0.0/esp_install-v5.0.0.zip
Downloading 0% 10% 20% 30% 40% 50% 60%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tl-install@5.0.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!
```

I then closed VSC and reopened it, and loaded the project, did a clean and then a build, and got this...

```
Processing rmp_release (platform: https://github.com/pioarduino/platform-espressif32/releases/download/54.03.21/platform-espressif32.zip; board: um_tinys2; framework: arduino)
--------------------------------------------------------------------------------------------------------------------------------------
idf_tools.py installation failed
idf_tools.py installation failed
Verbose mode can be enabled via `-v, --verbose` option
idf_tools.py installation failed
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/um_tinys2.html
PLATFORM: Espressif 32 (54.3.21) > Unexpected Maker TinyS2
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - tool-esptoolpy @ 5.0.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 58 compatible libraries
Scanning dependencies...
Dependency Graph
|-- Wire @ 3.2.1
|-- OneButton @ 2.6.1
|-- Adafruit BusIO @ 1.17.2
|-- Adafruit GFX Library @ 1.12.1
|-- Adafruit ILI9341 @ 1.6.2
|-- Adafruit BME280 Library @ 2.3.0
|-- PID @ 1.2.1
|-- PubSubClient @ 2.8.0
|-- AsyncTCP @ 3.4.5
|-- ESPAsyncWebServer @ 3.7.10
|-- SPI @ 3.2.1
|-- WiFi @ 3.2.1
|-- ESPmDNS @ 3.2.1
|-- HTTPClient @ 3.2.1
|-- FS @ 3.2.1
|-- SD @ 3.2.1
|-- FXL6408
|-- Adafruit Unified Sensor @ 1.1.15
|-- DNSServer @ 3.2.1
Building in release mode
Compiling .pio/build/rmp_release/libd2a/Wire/Wire.cpp.o
sh: xtensa-esp32s2-elf-g++: command not found
Compiling .pio/build/rmp_release/lib4a0/OneButton/OneButton.cpp.o
*** [.pio/build/rmp_release/libd2a/Wire/Wire.cpp.o] Error 127
sh: xtensa-esp32s2-elf-g++: command not found
*** [.pio/build/rmp_release/lib4a0/OneButton/OneButton.cpp.o] Error 127
===================================================== [FAILED] Took 1.39 seconds =====================================================

Environment    Status    Duration
-------------  --------  ------------
rmp_release    FAILED    00:00:01.392
=============================================== 1 failed, 0 succeeded in 00:00:01.392 ===============================================

 *  The terminal process "platformio 'run', '--environment', 'rmp_release'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

Please re-open this as this is not resolved.

Jason2866, 2025-07-27:
As long you see `idf_tools.py installation failed` there is a certificate problem.

UnexpectedMaker, 2025-07-27:
Your being intentionally cryptic and obtuse - I get you are an unpaid contributor and I do appreciate all of your hard work on this project, but being a smart arse doesn't help, it just puts me and other people off.

I followed your instructions - I extracted the 2 certs (CA and G2) and I installed them in KA, then set them to always trust.

I then re-started my Mac, and started VSC and tried to use 3.3.0 again and look and what I got.

[image]

UnexpectedMaker, 2025-07-27:
If I delete the `tool-esptoolpy` folder and then try to build, I see this...

[image]

but then it fails with...

[image]

and if I look at the folder, I see this...

[image]

and this is with this code in `idf_tools.py`

[image]

UnexpectedMaker, 2025-07-27:
error: /Users/seon/.platformio/packages/tool-esptoolpy does not appear to be a Python project, as neither `pyproject.toml` nor `setup.py` are present in the directory
Warning: Failed to install esptool: Command '['/Users/seon/.platformio/penv/bin/uv', 'pip', 'install', '--quiet', '--python=/Users/seon/.platformio/penv/bin/python', '-e', '/Users/seon/.platformio/packages/tool-esptoolpy']' returned non-zero exit status 2.

UnexpectedMaker, 2025-07-28:
Yes, python is brew installed, and xz is also installed. uv is also pip installed already

```
seon@seons-m2-max-mbp .platformio % source penv/bin/activate
(penv) seon@seons-m2-max-mbp .platformio % pip install uv
Requirement already satisfied: uv in ./penv/lib/python3.11/site-packages (0.8.3)
```

As I said, every pioarduino version works, even 3.2.1 did, until I tried 3.3.0 - now 3.2.1 and 3.3.0 both fail to install in various ways, but 3.1.3, 3.1.1, 3.0.7 all install and work fine.

Jason2866, 2025-07-28:
with 3.3.0 there are needed changes under the hood. Since you have all requirements installed which maybe can fail to auto install, i am honestly out of ideas. Do you have the folder dist in folder .platformio

TD-er, 2025-07-28:
I have never used any Apple device before, so maybe this doesn't make any sense...

Is it possible there is some tool active in the background which tries to manage Python virtual envs?
The reason I mention this is because someone reported a very odd issue with PyCharm constantly starting while building.
This was using Windows by the way.

In Windows (using WSL2) I had a lot of issues with installing esptool as I didn't have `uv` installed in the same virtual env as what VS-code uses.
I fixed this by opening a terminal window in VS code so I use the same virtual env as VS code is using and in there I installed `uv`.

Also installing esptool or switching between build envs which use different esptool (e.g. for ESP32 and ESP8266) may fail if there was still a 'monitoring' session open.
Most likely the renaming of folders already failed and the error was reported at the wrong moment.

UnexpectedMaker, 2025-08-02:
I'm still stuck on this - and it's unresolved, and the case is still closed, when clearly the certificates did not fix the issue for me.

Please re-open this so others can easily find it and possibly help.

UnexpectedMaker, 2025-08-03:
```
/Users/seon/.platformio/penv/bin/python /Users/seon/.platformio/packages/tl-install/tools/idf_tools.py --tools-json /Users/seon/.platformio/packages/tool-esptoolpy/tools.json install   

ERROR: It is not possible to determine the IDF version: [Errno 2] No such file or directory: '/Users/seon/esp-idf/esp_idf_version.h'
```

Why is the `.pioarduino` installer and tools using the IDF_PATH in my system and not the path inside the `.pioarduino` folder.

This clearly looks like a bug.  PIO + ESP32 + pioarduino should be able to work with or without ESP-IDF installed and should never be using an EDP-IDF path outside of the `.pioarduino` folder.

UnexpectedMaker, 2025-08-03:
@Jason2866 I've just proven this is a `pioarduino` build/install script issue:

If I remove the IDF_PATH from my `.zprofile` and re-boot, both 3.2.1 and 3.3.0 both install flawlessly - without any idf_tools.py errors.

If I put the path back in, both 3.2.1 and 3.3.0 fail to install - each at different places, as they are both checking the wrong requirements for the install because they are incorrectly using a path to an ESP-IDF install that has nothing to do with `pioarduino`

Also.... this is not an issue with 3.1.3 other 3.1.1 - so changes made since are causing this issue.

Also.... not a certificate issue, as I said from the very start.

I've lost weeks on this issue and have been unable to use platformio AT ALL on one of my Macs for over a week - and it would have been great if instead of ignoring me and this issue - you'd have worked with me to find this sooner.

Jason2866, 2025-08-03:
Forgot to ask to try with develop. Solved with https://github.com/pioarduino/platform-espressif32/pull/236

rac146, 2025-08-04:
I've been trying to get 3.2.X and 3.3.X to work as well on a Mac, and can say I am experiencing the same behavior as @UnexpectedMaker. It also seems when I make a switch from a working version (3.1.3) to one of the broken versions, I can first wipe ./platformio/packages (expecting a refresh with newest packages), but certain folders (like tool-cmake) end up with missing files after the update to 3.3.X (screenshot below). If I switch back to a 3.1.3 version, everything comes back just fine.

[image]

[image]

Would like to help find a fix, I just don't know where to begin - there's a lot of magic happening in the background.

Jason2866, 2025-08-05:
The original issue is solved in actual release. So the issues is closed. MacOS can make trouble with certificates. It is working.
I do the development on a Mac ;-) Some tips to solve the issues

# macOS Certificate Issues in Python Downloads - Analysis and Solutions

## Common Causes of Certificate Problems on macOS:

### 1. **Outdated Python Certificates**
- Python uses the `certifi` package for SSL certificates
- These can become outdated, especially with older Python installations
- **Solution**: `python3 -m pip install --upgrade certifi`

### 2. **macOS System Python (/usr/bin/python3) Issues**
- Apple's standard Python uses system certificates from the Keychain
- System updates can sometimes damage certificates
- **Solution**: Export system certificates and set as SSL_CERT_FILE

### 3. **macOS Keychain Problems** 
- macOS manages certificates through the Keychain
- System updates can sometimes corrupt certificates
- **Solution**: Keychain Access → Certificate Assistant → Evaluate

### 4. **Python.org Certificates Not Installed**
- With Python.org installations, certificates are often not installed automatically
- **Solution**: Run `/Applications/Python\ 3.x/Install\ Certificates.command`

### 5. **Corporate/Proxy Environment**
- Corporate proxies with their own certificates
- **Solution**: Add corporate CA certificates to Python certificate store

## Standard macOS Python (/usr/bin/python3) Solutions:

```bash
# 1. Export system certificates
security find-certificate -a -p /System/Library/Keychains/SystemRootCertificates.keychain > /tmp/system_certs.pem

# 2. Set environment variable
export SSL_CERT_FILE=/tmp/system_certs.pem

# 3. Update Python certifi for system Python
/usr/bin/python3 -m pip install --user --upgrade certifi

# 4. Update the system
# System Settings → General → Software Update

# 5. Test connection
curl -v https://dl.espressif.com
```

## Homebrew Python Solutions:

```bash
# Install Homebrew CA certificates
brew install ca-certificates

# Update Python certifi via Homebrew
brew reinstall python-certifi

# Test Homebrew Python
/opt/homebrew/bin/python3 -c "import ssl; print(ssl.get_default_verify_paths())"
```

## Python.org Installation Solutions:

```bash
# Run Install Certificates Command (if available)
/Applications/Python\ 3.11/Install\ Certificates.command
# or
/Applications/Python\ 3.12/Install\ Certificates.command

# Manually update certifi
python3 -m pip install --upgrade certifi
```

## Debugging Commands for macOS:

```bash
# Display Python SSL information
python3 -c "import ssl, certifi; print('SSL:', ssl.OPENSSL_VERSION); print('Certifi:', certifi.where())"

# Update certificates
python3 -m pip install --upgrade certifi

# Install Python.app certificates (if available)
/Applications/Python\ 3.*/Install\ Certificates.command

# Homebrew CA certificates
brew install ca-certificates

# Test HTTPS connection
python3 -c "import urllib.request; urllib.request.urlopen('https://dl.espressif.com')"
```

## Additional Solution Approaches:

### Set Environment Variables:
```bash
export SSL_CERT_FILE=$(python3 -m certifi)
export REQUESTS_CA_BUNDLE=$(python3 -m certifi)
```

### Bypass Corporate Proxy:
```bash
export https_proxy=""
export HTTPS_PROXY=""
```

### Manually Download Certificates:
```bash
# Manually add DigiCert Root CA
curl -O https://cacerts.digicert.com/DigiCertGlobalRootG2.crt
```

### Reset Certificate Cache:
```bash
# Clear pip and certificate caches
sudo rm -rf /tmp/pip-* && rm -rf ~/.cache/pip
```

## Python Installation Type Detection:

Different Python installations on macOS require different approaches:

### System Python (/usr/bin/python3):
- Uses macOS system certificates
- Requires system certificate export
- Best approach: Extract system certificates and set SSL_CERT_FILE

### Homebrew Python (/opt/homebrew or /usr/local):
- Uses Homebrew-managed certificates
- Best approach: Update via brew install ca-certificates

### Python.org Installation (/Applications/Python):
- Often missing certificates initially
- Best approach: Run Install Certificates.command script

## Testing Your Solution:

```bash
# Test 1: Basic SSL connection
python3 -c "import ssl; ssl.create_default_context().check_hostname = True"

# Test 2: Download test
python3 -c "import urllib.request; print('Success:', urllib.request.urlopen('https://dl.espressif.com').status)"

# Test 3: Certificate path verification
python3 -c "import certifi; print('Certifi path:', certifi.where())"

# Test 4: System certificate access
security find-certificate -a -p /System/Library/Keychains/SystemRootCertificates.keychain | head -20
```

rac146, 2025-08-06:
Alright I tried develop and I got this error:
```
Resolving xiao_esp32c6 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing xiao_esp32c6 (platform: https://github.com/pioarduino/platform-espressif32.git#develop; board: xiao_esp32c6; framework: espidf, arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/xiao_esp32c6.html
PLATFORM: Espressif 32 (55.3.30+develop.sha.e1e4448) > Xiao ESP32C6
HARDWARE: ESP32C6 160MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - contrib-piohome @ 3.4.4 
 - framework-arduinoespressif32 @ 3.3.0 
 - framework-arduinoespressif32-libs @ 5.5.0+sha.b66b5448e0 
 - framework-espidf @ 3.50500.0 (5.5.0) 
 - tool-cmake @ 3.30.2 
 - tool-esp-rom-elfs @ 2024.10.11 
 - tool-esptoolpy @ 5.0.2 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.13.1 
 - tool-scons @ 4.40801.0 (4.8.1) 
 - toolchain-riscv32-esp @ 14.2.0+20241119
Warning! Arduino framework as an ESP-IDF component doesn't handle the `variant` field! The default `esp32` variant will be used.
Reading CMake configuration...
-- Found Git: /usr/bin/git (found version "2.39.3 (Apple Git-146)")
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Minimal build - OFF
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/rac146/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /Users/rac146/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc
-- Check for working C compiler: /Users/rac146/.platformio/packages/toolchain-riscv32-esp/bin/riscv32-esp-elf-gcc - works
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Configuring incomplete, errors occurred!

fatal: not a git repository (or any of the parent directories): .git
CMake Error at /Users/rac146/Projects/remote/remote/.pio/build/xiao_esp32c6/CMakeFiles/3.30.2/CMakeSystem.cmake:6 (include):
  include could not find requested file:

    /home/vscode/.platformio/packages/framework-espidf/tools/cmake/toolchain-esp32c6.cmake
Call Stack (most recent call first):
  /Users/rac146/Projects/remote/remote/.pio/build/xiao_esp32c6/CMakeFiles/CMakeScratch/TryCompile-YV6lCl/CMakeLists.txt:5 (project)

CMake Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
CMake Error at /Users/rac146/.platformio/packages/tool-cmake/share/cmake-3.30/Modules/CMakeDetermineCompilerABI.cmake:74 (try_compile):
  Failed to configure test project build system.
Call Stack (most recent call first):
  /Users/rac146/.platformio/packages/tool-cmake/share/cmake-3.30/Modules/CMakeTestCXXCompiler.cmake:26 (CMAKE_DETERMINE_COMPILER_ABI)
  /Users/rac146/.platformio/packages/framework-espidf/tools/cmake/project.cmake:589 (__project)
  CMakeLists.txt:3 (project)

========================== [FAILED] Took 4.92 seconds ==========================

```

Build throws this error:

[image]

Hope that helps..

rac146, 2025-08-06:
But there isn't. As I stated earlier this same exact project works and loads perfectly fine in both windows and my Linux container. Pretty frustrating - I'm trying to supply as much info as possible to track this down.

rac146, 2025-08-06:
OK I got to a working state!! I completely wiped the .platformio folder and started a project from scratch with your develop branch - this works!

I would suggest to others experiencing a similar issue to wipe the .platformio folder and starting over. Sorry for the doubt.

doudar, 2025-08-11:
@Jason2866 , I've tried everything in this thread (including certificates as well as the last post about nuking the .platformio folder) and the only fix for me was urllib.

Jason2866, 2025-08-25:
@45gfg9 Can you try if this version works https://github.com/pioarduino/esp_install/blob/certi_search/tools/idf_tools.py
Replace only the file and delete some stuff in folder `dist` If it works it downloads the missing packages

45gfg9, 2025-08-26:
> [@45gfg9](https://github.com/45gfg9) Can you try if this version works [pioarduino/esp_install@`certi_search`/tools/idf_tools.py](https://github.com/pioarduino/esp_install/blob/certi_search/tools/idf_tools.py?rgh-link-date=2025-08-25T23%3A14%3A20.000Z) Replace only the file and delete some stuff in folder `dist` If it works it downloads the missing packages

Yes, I can confirm this version works for me.

Jason2866, 2025-08-26:
@45gfg9 Can you try branch develop. I have added certifi in the penv (venv) and it is setting the cert path.
https://github.com/pioarduino/platform-espressif32/blob/b528557b69d335e9edd4e1d876daed2cb1d3227e/builder/penv_setup.py#L448-L469

Xylopyrographer, 2025-08-26:
@Jason2866 

Also having install problems on my Mac. Regarding the solutions above, worth noting the differences between the hardware:

On a macOS 15 system, the default Homebrew installation location depends on your Mac's processor type:

`/opt/homebrew` for Apple Silicon (M1/M2/M3/M4) Macs and
`/usr/local` for older Intel Macs. 

All packages are installed within this designated directory, with their binaries often linked into `/opt/homebrew/bin` or `/usr/local/bin` for easy access.

For Apple Silicon (M1/M2/M3/M4) Macs: 

- Installation prefix: `/opt/homebrew`.
- Binaries: Located in `/opt/homebrew/bin`.
- Packages: Stored in their own subdirectories under `/opt/homebrew/Cellar`.

For Intel Macs: 

- Installation prefix: `/usr/local`.
- Binaries: Placed in `/usr/local/bin`.
- Packages: Stored in their own subdirectories under `/usr/local/Cellar`.

To confirm the location on your system:

You can try running the Homebrew help command from each location. One of the following commands should work, depending on your Mac's architecture:

`/usr/local/bin/brew help` (for Intel Macs)
`/opt/homebrew/bin/brew help` (for Apple Silicon Macs)

So this would impact the great [reference above](https://github.com/pioarduino/platform-espressif32/issues/235#issuecomment-3155742847).

But might it also affect the installer if checks aren't being made for the undelying machine architecture?

FWIW, I'm running a 2020 Intel based iMac. Still working on resolving the instll issues.

Xylopyrographer, 2025-08-26:
Quick follow-up: On my Intel based Mac, running:

`/usr/local/bin/python3 -c "import ssl; print(ssl.get_default_verify_paths())"`

returns:

`DefaultVerifyPaths(cafile='/usr/local/etc/openssl@3/cert.pem', capath='/usr/local/etc/openssl@3/certs', openssl_cafile_env='SSL_CERT_FILE', openssl_cafile='/usr/local/etc/openssl@3/cert.pem', openssl_capath_env='SSL_CERT_DIR', openssl_capath='/usr/local/etc/openssl@3/certs')`

Jason2866, 2025-08-27:
If still issues exist i am completely out of ideas. In this case i need full access to the problematic PC.

Migsi, 2025-09-21:
It appears the script has a soft dependency on python-certifi, that should actually be a hard one. Just encountered the same issue on amd64 Linux and installing that package system wide solved the issue.

Jason2866, 2025-09-21:
@Migsi  Can you try branch:
https://github.com/pioarduino/platform-espressif32/tree/mv_py_setup
It now installs the penv before calling the installer from the penv. This makes sure that a defined Python env is used for the Installer. No more dependency at all to the System Python now.

Xylopyrographer, 2025-10-30:
Leaving a footnote here for posterity and should someone else need to see it...

I took another step back to try and understand causes of the issues I've was having. Long story short, over the years, turns out the system $PATH was in a state of complete disarray. One of the consequences was a very detrimental effect it was having as well on the system Python environment. Since setting those two things right, there have been no issues with pioarduino.

Might be something to investigate if you are having issues.