pioarduino/platform-espressif32 #239: OpenOCD does not get installed: debug not working

jjchico, 2025-07-31:
### PROBLEM DESCRIPTION
I am trying to debug a project for a esp32 super mini board from VSCode. Build and upload works right. Debug build works right but the debugger does not start because the openocd tool is not installed. The debug console says:

Unpacking...
Tool Manager: tool-openocd-esp32@2.1200.20250707 has been installed!
Warning! Please install `99-platformio-udev.rules`. 
More details: https://docs.platformio.org/en/latest/core/installation/udev-rules.html
undefinedDebugInvalidOptionsError: Could not launch Debug Server '/home/jjchico/.platformio/packages/tool-openocd-esp32/bin/openocd'. Please check that it is installed and is included in a system PATH
See https://docs.platformio.org/page/plus/debugging.html

If the debug process is executed again, the "Unpacking..." and "... has been installed" lines are not there any more but the same error occurs. It looks like OpenOCD metadata is installed but not the actual executable.

### TO REPRODUCE
1. I create a new default Arduino project for the Nologo ESP32 Super Mini.
2. Build and upload the project.
3. Execute a PIO debugger session.
4. Look at the output in the debug console.

### EXPECTED BEHAVIOUR
The debugger session starts.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

=== platformio.ini
----
[platformio]
default_envs = pioarduino-devel

[env]
board = nologo_esp32c3_super_mini
framework = arduino

[env:pioarduino]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip

[env:pioarduino-devel]
platform = https://github.com/pioarduino/platform-espressif32.git#develop
---

=== OpenOCD tool folder after executing the debug process

$ ls -l ~/.platformio/packages/tool-openocd-esp32/
total 8
-rw-rw-r-- 1 jjchico jjchico  420 jul  7 13:40 package.json
-rw-rw-r-- 1 jjchico jjchico 3485 jul  7 13:39 tools.json

**(Please, remember to close the issue when the problem has been addressed)**

jjchico, 2025-07-31:
> Seems using Linux. Which distribution and version. Looks like a not full Python version. Do `sudo apt install python3-full` and tar needs xz support do `sudo apt install xz-utils`

On Ubuntu 24.04. Installed `python3-full` and `xz-utils` but there is no difference.

What I see is that pioarduino looks at platforms/espressif32/platforms.json where it says:

    "tool-openocd-esp32": {
      "type": "debugger",
      "optional": true,
      "owner": "pioarduino",
      "package-version": "2.1200.20250707",
      "version": "https://github.com/pioarduino/registry/releases/download/0.0.1/openocd-v0.12.0-esp32-20250707.zip"

The zip is installed but it only contains:

    $ ls -A packages/tool-openocd-esp32/
    package.json  .piopm  tools.json

The tools.json file includes the link to the binary tool but it is never downloaded/installed and no error is triggered.

    "versions": [
        {
          "linux-amd64": {
            "sha256": "766293bd7a08900d3536f87a0a7ade960f07266f16e4147f95ca5ce4e15d4c5d",
            "size": 2489724,
            "url": "https://github.com/espressif/openocd-esp32/releases/download/v0.12.0-esp32-20250707/openocd-esp32-linux-amd64-0.12.0-esp32-20250707.tar.gz"

I have tried to download and copy the .tar.gz to the tool folder, but now the debugger complains about missing GDB. It is not even trying to download and install it even if the information in platforms.json looks right (I can download the blop manually).

Jason2866, 2025-08-02:
Try #241