pioarduino/platform-espressif32 #244: error while compile, no such file or directory with build-in header-files

rk-sTEk, 2025-08-03:
### PROBLEM DESCRIPTION
Compiling error - unable to find header-files,but them are in the framework

### TO REPRODUCE
one of the error-Messages:

Compiling .pio/build/esp32-c6-devkitm-1/lib0a9/WebServer/middleware/CorsMiddleware.cpp.o
/home/.../.platformio/packages/framework-arduinoespressif32/libraries/WebServer/src/Parsing.cpp:24:10: fatal error: NetworkServer.h: No such file or directory

***********************************************************************
* Looking for NetworkServer.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:NetworkServer.h"
* Web  > https://registry.platformio.org/search?q=header:NetworkServer.h
*
***********************************************************************

   24 | #include "NetworkServer.h"
      |          ^~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pio/build/esp32-c6-devkitm-1/lib0a9/WebServer/Parsing.cpp.o] Error 1

### EXPECTED BEHAVIOUR
My platformio.ini is:

[code]
[env:esp32-c6-devkitm-1]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip

lib_compat_mode = strict
lib_ldf_mode = chain
board = esp32-c6-devkitm-1
framework = arduino
build_flags = 
	-D ARDUINO_USB_MODE=1 
	-D ARDUINO_USB_CDC_ON_BOOT=1
        -D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000
        -D CONFIG_ASYNC_TCP_PRIORITY=10
        -D CONFIG_ASYNC_TCP_QUEUE_SIZE=64
        -D CONFIG_ASYNC_TCP_RUNNING_CORE=1
        -D CONFIG_ASYNC_TCP_STACK_SIZE=4096

lib_deps = 
	adafruit/Adafruit ST7735 and ST7789 Library @ ^1.11.0
	ESP32Async/ESPAsyncWebServer
	ESP32Async/AsyncTCP
	bblanchon/ArduinoJson@^7.4.2
monitor_speed = 115200

[/code]

### SCREENSHOTS

### ADDITIONAL CONTEXT
I looked for the files, they are in the framework directrory.

Jason2866, 2025-08-03:
Not a pioarduino issue, probably caused by one of the lib_deps. Try to find out which one causes the problem and open and issue in the repo of the maintainer of the lib

rk-sTEk, 2025-08-03:
Thxfor your answer. The spaces were lost by copy+paste.