pioarduino/platform-espressif32 #248: Stable Arduino is not based on Arduino 3.3.0 and IDF 5.5.0

xgnata, 2025-08-06:
### PROBLEM DESCRIPTION
Stable Arduino is not based on Arduino 3.3.0 and IDF 5.5.0  as stated in https://github.com/pioarduino/platform-espressif32?tab=readme-ov-file
Any project compiled with https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip shows:
 framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - tool-esptoolpy @ 5.0.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119

### TO REPRODUCE
Use 
[env:stable]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
platformio.ini
The compilation process reports:
 framework-arduinoespressif32 @ 3.2.1 
 - framework-arduinoespressif32-libs @ 5.4.0+sha.858a988d6e 
 - tool-esptoolpy @ 5.0.0 
 - toolchain-xtensa-esp-elf @ 14.2.0+20241119

### EXPECTED BEHAVIOUR
Stable Arduino is not based on Arduino 3.3.0 and IDF 5.5.0

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-08-06:
Delete the hidden folder `.platformio/platforms` this will download latest stable version -> Arduino core 3.3.0

xgnata, 2025-08-07:
I'm missing something. Sorry if I'm dumb...
I did wipe out .platformio/platforms but when I re-build the project, it still resolves/downloads  esp32-3.2.1.zip and esp32-arduino-libs-idf-release_v5.4-858a988d-v1.zip
 
Resolving stable dependencies...
Platform Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Platform Manager: espressif32@54.3.21 has been installed!
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.2.1/esp32-3.2.1.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32@3.2.1 has been installed!
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.4/esp32-arduino-libs-idf-release_v5.4-858a988d-v1.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-arduinoespressif32-libs@5.4.0+sha.858a988d6e has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/xtensa-esp-elf-14.2.0_20241119.zip
Unpacking 0% 10% 20%
Tool Manager: toolchain-xtensa-esp-elf@14.2.0+20241119 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esptool/releases/download/v5.0.0-dev1/esptool.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esptoolpy@5.0.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp_install/releases/download/v5.0.0/esp_install-v5.0.0.zip
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tl-install@5.0.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!