pioarduino/platform-espressif32 #259: Esp32 C2 seems busted

zackees, 2025-08-13:
Fastled seems to have a regression with the esp32c2

This platformio.ini

```
[env:esp32-c2-devkitm-1]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino
board = esp32-c2-devkitm-1
monitor_speed = 115200
```

```
$ pio run
Processing esp32-c2-devkitm-1 (platform: https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip; framework: arduino; board: esp32-c2-devkitm-1)
-------------------------------------------------------------------------------------------------------idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
Verbose mode can be enabled via `-v, --verbose` option
idf_tools.py installation failed
idf_tools.py installation failed
idf_tools.py installation failed
PYTHONEXE updated to penv environment: C:\Users\niteris\.platformio\penv\Scripts\python.exe
error: C:\Users\niteris\.platformio\packages\tool-esptoolpy does not appear to be a Python project, as 
neither `pyproject.toml` nor `setup.py` are present in the directory
Warning: Failed to install esptool: Command '['C:\\Users\\niteris\\.platformio\\penv\\Scripts\\uv.exe', 'pip', 'install', '--quiet', '--python=C:\\Users\\niteris\\.platformio\\penv\\Scripts\\python.exe', '-e', 'C:\\Users\\niteris\\.platformio\\packages\\tool-esptoolpy']' returned non-zero exit status 2.     
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c2-devkitm-1.html
PLATFORM: Espressif 32 (55.3.30) > Espressif ESP32-C2-DevKitM-1
HARDWARE: ESP32C2 120MHz, 272KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, 
olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - contrib-piohome @ 3.4.4
 - framework-arduinoespressif32 @ 3.3.0
 - framework-arduinoespressif32-libs @ 5.5.0+sha.b66b5448e0
 - tool-esptoolpy @ 5.0.2
 - tool-mklittlefs @ 3.2.0
 - toolchain-riscv32-esp @ 14.2.0+20241119
*** Compile Arduino IDF libs for esp32-c2-devkitm-1 ***
AttributeError: 'NoneType' object has no attribute 'split':
  File "C:\Users\niteris\dev\fastled2\.venv\Lib\site-packages\platformio\builder\main.py", line 173:   
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\main.py", line 1039:
    target_elf = env.BuildProgram()
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\niteris\dev\fastled2\.venv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 62:
    env.ProcessProgramDeps()
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\niteris\dev\fastled2\.venv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 142:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Util\envs.py", line 252:
    return self.method(*nargs, **kwargs)
  File "C:\Users\niteris\dev\fastled2\.venv\Lib\site-packages\platformio\builder\tools\piobuild.py", line 352:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 684:
    return method(*args, **kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 890:   
    call_compile_libs()
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 640:   
    SConscript("espidf.py")
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 684:
    return method(*args, **kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 76:     
    .split(".")[1]
===================================== [FAILED] Took 4.46 seconds =====================================
(fastled2) 
```

Jason2866, 2025-08-13:
idf is not installed. Please post your used boards.json? Seem like it is not the one from the repo.
Probaby missing the entry:
```
  "espidf": {
    "custom_sdkconfig": [
      "CONFIG_IDF_TARGET=\"esp32c2\"" 
    ]
  },
```

Jason2866, 2025-08-13:
For C2 "somewhere" the setting `CONFIG_IDF_TARGET="esp32c2"` is needed to trigger HybridCompile.

zackees, 2025-08-13:
It does not fix it:

**platformio.ini**
```
[env:esp32c2]
board = esp32-c2-devkitm-1
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
framework = arduino
build_flags =
    -DCONFIG_IDF_TARGET_ESP32C2=1
    -Isrc/sketch
    -fopt-info-all=C:/Users/niteris/dev/fastled2/.build/pio/esp32c2/optimization_report.txt
    -Wl,-Map,firmware.map
custom_sdkconfig = CONFIG_IDF_TARGET=esp32c2
lib_ldf_mode = chain
lib_archive = true
``

**pio run**
```
Zach Vorhies@DESKTOP-I3718DO MINGW64 ~/dev/fastled2/.build/pio/esp32c2 (master)
$ pio run
Processing esp32c2 (board: esp32-c2-devkitm-1; platform: https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip; framework: arduino)  
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing https://github.com/pioarduino/esp_install/releases/download/v5.1.0/esp_install-v5.1.0.zip
Unpacking  [####################################]  100%
Tool Manager: tool-esp_install@5.1.0 has been installed!
Tool Manager: Installing https://github.com/espressif/arduino-esp32/releases/download/3.3.0/esp32-3.3.0.zip
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32@3.3.0 has been installed!
Tool Manager: Installing https://github.com/espressif/esp32-arduino-lib-builder/releases/download/idf-release_v5.5/esp32-arduino-libs-idf-release_v5.5-b66b5448-v1.zip
Unpacking  [####################################]  100%
Tool Manager: framework-arduinoespressif32-libs@5.5.0+sha.b66b5448e0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-alpha1/c2_cbe9388f45_compile_skeleton.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-arduino-c2-skeleton-lib@5.5.0+sha.cbe9388f45 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.5.0/esp-idf-v5.5.0.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: framework-espidf@3.50500.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/riscv32-esp-elf-14.2.0_20241119.zip
Unpacking  [####################################]  100%
Tool Manager: toolchain-riscv32-esp@14.2.0+20241119 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esptoolpy-v5.0.2.zip
Unpacking  [####################################]  100%
Tool Manager: tool-esptoolpy@5.0.2 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/contrib-piohome-3.4.4.tar.gz
Unpacking  [####################################]  100%
Tool Manager: contrib-piohome@3.4.4 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/mklittlefs-3.2.0-new.zip
Unpacking  [####################################]  100%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/cmake-3.30.2.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-cmake@3.30.2 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esp-rom-elfs-20241011.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/ninja-1.13.1.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-ninja@1.13.1 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/scons-4.8.1.zip
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Unpacking  [####################################]  100%
[... 110 lines cut ...]
    call_compile_libs()
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\frameworks\arduino.py", line 640:
    SConscript("espidf.py")
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 684:
    return method(*args, **kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 620:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\niteris\.platformio\packages\tool-scons\scons-local-4.8.1\SCons\Script\SConscript.py", line 280:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1795:
    project_codemodel = get_cmake_code_model(
  File "C:\Users\niteris\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 490:
    run_cmake(src_dir, build_dir, extra_args)
[... cut ...]
  File "C:\Users\niteris\AppData\Roaming\uv\python\cpython-3.12.6-windows-x86_64-none\Lib\subprocess.py", line 1026:
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\niteris\AppData\Roaming\uv\python\cpython-3.12.6-windows-x86_64-none\Lib\subprocess.py", line 1538:
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
=========================================================================== [FAILED] Took 201.38 seconds ===========================================================================(ingest-media) 
```

Jason2866, 2025-08-13:
You need to find the reason why the installer fails. There something specific in your setup.

TD-er, 2025-08-14:
@Jason2866 Is that a similar timeout as you saw in my runs lately regarding DNS or is this a different timeout?

zackees, 2025-08-14:
Timeouts are necessary and evil at the same time.

FastLED has a [special class for that solves it for us](https://github.com/FastLED/FastLED/blob/fe3452ea8e81083dec7dff7705d1ef40504af2ea/ci/util/running_process.py#L233) that we use for all our tests.

Jason2866, 2025-08-14:
A timeout of 300 seconds is for sure long enough. There is something badly wrong with the Internet connection.

Jason2866, 2025-08-14:
Closing since not an issue of pioarduino

TD-er, 2025-08-14:
Just an idea...
Referring to an online repo is an instruction to download.
But what if you refer not to an online repo but a local clone of the same repo?

zackees, 2025-08-14:
> Closing since not an issue of pioarduino

How is this not an issue? I'm not running this offline, i'm running this from my windows desktop in the Mission District in San Francisco.