pioarduino/platform-espressif32 #260: pio check not working

SinglWolf, 2025-08-14:
### PROBLEM DESCRIPTION

Checking code not working...
OS `LINUX`, `WINDOWS` no difference.
### TO REPRODUCE
_Steps to reproduce the behavior:_

```bash
pio check -e S3_ILI9341_VS
Checking S3_ILI9341_VS > cppcheck (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip; framework: arduino, espidf)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
TypeError: Traceback (most recent call last):
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/check/cli.py", line 156, in cli
    rc = ct.check(
         ^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/check/tools/cppcheck.py", line 260, in check
    cmd = self.configure_command(scope, src_file)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/check/tools/cppcheck.py", line 106, in configure_command
    tool_path = os.path.join(self.get_tool_dir("tool-cppcheck"), "cppcheck")
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen posixpath>", line 76, in join
TypeError: expected str, bytes or os.PathLike object, not NoneType

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```
It's the same through the GUI

[image]

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT

I need this particular version of `platform-espressif32` due to the limitations of the `ESPAsyncWebServer` library.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-08-14:
Please try latest release, changes have been made since the version you use.

Jason2866, 2025-08-14:
ESPAsyncWebServer does support latest Arduino release https://github.com/ESP32Async/ESPAsyncWebServer

SinglWolf, 2025-08-14:
In file https://github.com/ESP32Async/ESPAsyncWebServer/blob/main/idf_component.yml there is a clear indication of the arduino version - 3.1.1
```yml
dependencies:
  espressif/arduino-esp32:
    version: "^3.1.1"
```

And yes, I tried compiling the library on the latest version. It does not compile.

Jason2866, 2025-08-14:
`version: "^3.1.1"` is fine all versions up to 3.9.9 are allowed

SinglWolf, 2025-08-14:
> Но, кажется, я нашёл проблему. Нужно внести изменения в ядро pioarduino. Перезапишите существующий файл в .platformio/penv в libsitepackes в Platformio в подкаталоге platformio/check/tools/ этим файлом https://github.com/Jason2866/platformio-core/blob/pioarduino/platformio/check/tools/cppcheck.py

Thanks! I'll try it in the morning. It's after midnight.

> `vscode-esp-idf-extension` is IDF

I'm aware. 😃

Jason2866, 2025-08-14:
It does compile, see example https://github.com/ESP32Async/ESPAsyncWebServer/blob/main/pioarduino_examples/IncreaseMaxSockets/platformio.ini
which uses HybridCompile mode from pioarduino (which is simple way to compile Arduino as an component of IDF)

SinglWolf, 2025-08-14:
> It does compile, see example https://github.com/ESP32Async/ESPAsyncWebServer/blob/main/pioarduino_examples/IncreaseMaxSockets/platformio.ini which uses HybridCompile mode from pioarduino (which is simple way to compile Arduino as an component of IDF)

Thanks! I'll try it in the morning.

SinglWolf, 2025-08-15:
> It does compile, see example https://github.com/ESP32Async/ESPAsyncWebServer/blob/main/pioarduino_examples/IncreaseMaxSockets/platformio.ini which uses HybridCompile mode from pioarduino (which is simple way to compile Arduino as an component of IDF)

The compilation is probably going well, but the `pio check` still doesn't work in `Windows 10`.

```cmd
PS C:\Users\Alexander\Documents\GitHub\Radiola-ex> pio check -e S3_ILI9341_VS
Checking S3_ILI9341_VS > cppcheck (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-2/platform-espressif32.zip; framework: arduino, espidf)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------TypeError: Traceback (most recent call last):
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ~~~^^
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\click\core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\click\core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\cli.py", line 85, in invoke
    return super().invoke(ctx)
           ~~~~~~~~~~~~~~^^^^^
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\click\core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\click\core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\click\core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\check\cli.py", line 156, in cli
    rc = ct.check(
        on_defect_callback=(
    ...<3 lines>...
        )
    )
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\check\tools\cppcheck.py", line 260, in check
    cmd = self.configure_command(scope, src_file)
  File "C:\Users\Alexander\.platformio\penv\Lib\site-packages\platformio\check\tools\cppcheck.py", line 106, in configure_command
    tool_path = os.path.join(self.get_tool_dir("tool-cppcheck"), "cppcheck")
  File "<frozen ntpath>", line 100, in join
TypeError: expected str, bytes or os.PathLike object, not NoneType

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

SinglWolf, 2025-08-15:
> But i think i found the problem, it will need changes in pioarduino core. Overwrite the existing file in .platformio/penv in libsitepackes in Platformio in a sub dir platformio/check/tools/ with this one https://github.com/Jason2866/platformio-core/blob/pioarduino/platformio/check/tools/cppcheck.py

Your Linux fix doesn't work.

```bash
alex@rosa-ha1ejs ~/eez-projects/Radiola-ex (main) $ pio check -e S3_ILI9341_VS
Checking S3_ILI9341_VS > cppcheck (board: ESP32-S3-DevKitC-1-N16R8; platform: https://github.com/pioarduino/platform-espressif32/releases/download/53.03.11/platform-espressif32.zip; framework: arduino, espidf)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
FileNotFoundError: Traceback (most recent call last):
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/__main__.py", line 103, in main
    cli()  # pylint: disable=no-value-for-parameter
    ^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/cli.py", line 85, in invoke
    return super().invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/check/cli.py", line 156, in cli
    rc = ct.check(
         ^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/check/tools/cppcheck.py", line 272, in check
    self.execute_check_cmd(cmd)
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/check/tools/base.py", line 192, in execute_check_cmd
    result = proc.exec_command(
             ^^^^^^^^^^^^^^^^^^
  File "/home/alex/.platformio/penv/lib64/python3.11/site-packages/platformio/proc.py", line 114, in exec_command
    with subprocess.Popen(*args, **kwargs) as p:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib64/python3.11/subprocess.py", line 1026, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib64/python3.11/subprocess.py", line 1955, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: '/home/alex/.platformio/packages/tool-cppcheck/cppcheck'

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `python -m pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq/index.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

 
```

Jason2866, 2025-08-15:
fixed by rerelease of pioarduino core v6.1.18 with https://github.com/pioarduino/platformio-core/commit/65af112b49225736b5d6c0dfb4645dac8c3b223d

to use delete folder `.platformio/penv` and all cache folders

closed 2025-08-15 by Jason2866