pioarduino/platform-espressif32 #262: Potential conflict with ESP-IDF env vars

kjkent, 2025-08-19:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

As I sometimes use ESP-IDF outside of platformio and pioarduino, I have the following environment variables set as I want to keep as many dotfiles/dotfolders outside of `$HOME`:

- `IDF_PATH='/home/kjkent/.local/share/esp/idf/v5.5/idf'
- `IDF_TOOLS_PATH='/home/kjkent/.local/share/esp/tools'

Since switching to pioarduino from platformio's `espressif32` platform, I've encountered the bug described in https://github.com/pioarduino/platform-espressif32/issues/208 (configuration and build failures including `idf_tools.py installation failed`, `xtensa-esp32-elf-g++: not found`). I had tried the suggestions in that issue; mainly deleting the contents of `PLATFORMIO_CORE_DIR`, `PLATFORMIO_PACKAGES_DIR`

Perhaps this is related to the existing environment host environment; so, I unset `IDF_PATH` and `IDF_TOOLS_PATH` and restarted VS Code (I am using PlatformIO IDE), and the toolchain installation completed successfully, including `tool-esptoolpy`.

### TO REPRODUCE

1. With the environment variables `IDF_PATH=$HOME/<path>` and `IDF_TOOLS_PATH=$HOME/<path>` exported, launch VS Code from a terminal using `code`. If VS Code is launched from its `.desktop` file, it inherits the systemd user activation environment, and so would need `dbus-update-activation-environment --systemd --all` following env var export to be picked up by VS Code.

2. Create a new project in PlatformIO IDE, with the following `platformio.ini`:

```ini
[env:freenove_esp32_s3_wroom]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/55.03.30-2/platform-espressif32.zip
framework = arduino
```

3. Attempt to build a project. In my case, I was attempting to build https://github.com/s60sc/ESP32-CAM_MJPEG2SD

4. Failures in installing the toolchain are observed, with `tool-esptoolpy` failing to install.

5. Restart VS Code without `IDF_TOOLS_PATH` and `IDF_PATH` in environment

6. Toolchain installs successfully on PlatformIO startup.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

PlatformIO / pioarduino checks for and unsets these environment variables within its scripts to avoid conflict with ESP-IDF installations and associated environment variables.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Machine specs:

```text
OS: Arch Linux x86_64
Kernel: Linux 6.16.1-zen1-1-zen
Shell: fish 4.0.2
WM: Hyprland 0.50.1 (Wayland)
Terminal: WezTerm 20250730-195751-6a493f88
CPU: AMD Ryzen 5 3600 (12) @ 4.21 GHz
GPU: NVIDIA GeForce RTX 3060 Ti [Discrete]
Locale: en_GB.UTF-8
```

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-08-19:
fixed in branch develop

Jason2866, 2025-08-19:
it is fixed since https://github.com/pioarduino/platform-espressif32/pull/236

kjkent, 2025-08-19:
@Jason2866 whoops, apologies for not spotting that! I'll close this now.