pioarduino/platform-espressif32 #264: The application for ESP32-P4 no longer compiles

cniedzi, 2025-08-21:
### PROBLEM DESCRIPTION
Board ESP32-P4-DevKitC-1-NRW32

My application for the ESP32-P4 module no longer compiles. Some time ago it built without any problems; I haven’t changed my configuration and I don’t know what caused the issue. Now I’m receiving the following error:

```
Compiling .pio\build\esp32p4\libdb1\WebServer\middleware\AuthenticationMiddleware.cpp.o
Compiling .pio\build\esp32p4\libdb1\WebServer\middleware\CorsMiddleware.cpp.o
In file included from C:/Users/Czarek/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.cpp:245:
managed_components/espressif__esp_hosted/host/api/include/esp_hosted_transport_config.h:12:10: fatal error: port_esp_hosted_host_config.h: No such file or directory
```

The file `port_esp_hosted_host_config.h` does exist in the `managed_components\espressif__esp_hosted\host\port\esp\freertos\include\` directory. 

When I add the full path `#include "..\..\port\esp\freertos\include\port_esp_hosted_host_config.h"` in the file `esp_hosted_transport_config.h`, the program compiles without any issues.

My `platformio.ini` file:

```
; PlatformIO Project Configuration File

[esp32p4]
    board_build.partitions = boards/partitions.csv   ; 16MB Flash
    board = ESP32-P4-DevKitC-1-NRW32 ; 16 MB Flash, has PSRAM

[env]
   platform = https://github.com/pioarduino/platform-espressif32.git

    monitor_speed   = 115200
    monitor_filters = esp32_exception_decoder
    monitor_rts     = 0
    monitor_dtr     = 0
    upload_speed    = 921600

[env:esp32p4]
    board = ${esp32p4.board}
    board_build.partitions = ${esp32p4.board_build.partitions}

    build_flags =
        -D CORE_DEBUG_LEVEL=3                    ; 0 None, 1 Error, 2 Warn, 3 Info, 4 Debug, 5 Verbose
        -D CONFIG_ARDUHAL_LOG_COLORS=1
        -D ARDUINO_RUNNING_CORE=1                ; Arduino Runs On Core (setup, loop)
        -D ARDUINO_EVENT_RUNNING_CORE=1          ; Events Run On Core
        -D BOARD_HAS_PSRAM=1
        -D MONITOR_SPEED=${env.monitor_speed}
        -D CONFIG_ESP32_PHY_MAX_WIFI_TX_POWER=20
        -D ESP32P4                
        -D ST7789                  

    lib_deps =
        https://github.com/schreibfaul1/ESP32-audioI2S.git
        https://github.com/ESP32Async/ESPAsyncWebServer
        https://github.com/tzapu/WiFiManager
        https://github.com/adafruit/Adafruit-GFX-Library

    framework = arduino, espidf
```

Do I do something wrong or should I add any configuration in platformio.ini?

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

cniedzi, 2025-08-22:
> Not a pioarduino issues. See [espressif/esp-hosted-mcu#96](https://github.com/espressif/esp-hosted-mcu/issues/96) Use a older version of the component until the bug in the component esp-hosted-mcu is solved.

@Jason2866, could you please advise how to use/switch to an older version of this component?

cniedzi, 2025-08-22:
> Search in `.platformio/package` folder the framework arduino folder and change in the file https://github.com/espressif/arduino-esp32/blob/956b687389a3cb63a6d1f0142fbbbe127f697db9/idf_component.yml#L110 the version to a fixed one. Try version `2.2.4`

Thank you! I've changed the content of the file `C:\Users\XXX\.platformio\packages\framework-arduinoespressif32\idf_component.yml` to: 
  ```
espressif/esp_hosted:
    version: "^2.2.4"
    rules:
      - if: "target == esp32p4"
```
but platformio includes the managed_component "espressif__esp_hosted" in version 2.3.0 which still doesn't work for me:
```
dependencies:
  idf:
    version: '>=5.3'
description: ESP-Hosted-MCU provide drivers to act any ESP chipset as Wi-Fi or Bluetooth
  co-processor.
repository: git://github.com/espressif/esp-hosted-mcu.git
repository_info:
  commit_sha: b76543512d028fc2534cf1700c611c335148986a
  path: .
url: https://github.com/espressif/esp-hosted-mcu
version: 2.3.0
```

Jason2866, 2025-08-22:
The v2.3.0 is the broken one. You have to use an earlier one by pinning it correctly:
```
espressif/esp_hosted:
  version: "==2.2.4"
  rules:
    - if: "target == esp32p4"
```

cniedzi, 2025-08-22:
> The v2.3.0 is the broken one. You have to use an earlier one by pinning it correctly:
> 
> ```
> espressif/esp_hosted:
>   version: "==2.2.4"
>   rules:
>     - if: "target == esp32p4"
> ```

Thanks! Works like a charm :)