pioarduino/platform-espressif32 #278: ESP32-S3 toolchain and idf_tools.py installation fails on MacBook M4 (Apple Silicon, ARM64)

pdo59, 2025-09-01:
### PROBLEM DESCRIPTION
On my MacBook M4 (Apple Silicon, ARM64), the ESP32-S3 toolchain and `idf_tools.py` installation fail when using the Waveshare PlatformIO platform. The same project works perfectly on my Intel iMac. Compilation cannot proceed because the toolchain is not installed and `xtensa-esp32s3-elf-gcc` is not found.

### TO REPRODUCE
1. Create a PlatformIO project for ESP32-S3 using:
   ```
   platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
   board = ESP32-S3-Touch-LCD-7B
   framework = arduino
   ```
2. Run `pio run` or `pio pkg install` in the project directory.

### EXPECTED BEHAVIOUR
The toolchain and all dependencies should install and compile successfully on Apple Silicon Macs, as they do on Intel Macs.

### SCREENSHOTS
Error messages:
```
idf_tools.py installation failed
error: .../tool-esptoolpy does not appear to be a Python project, as neither pyproject.toml nor setup.py are present in the directory
sh: xtensa-esp32s3-elf-gcc: command not found
```

### ADDITIONAL CONTEXT
- MacBook M4 (Apple Silicon, ARM64), macOS Sequoia 15.6.1
- Python 3.11.7, pip 25.2, PlatformIO Core 6.1.18·Home 3.4.4
- The official PlatformIO ESP32 platform (`platform = espressif32`) installs the toolchain correctly, but I need to use the Waveshare platform for my board.
- I have tried cleaning `.platformio`, updating Python and pip, and reinstalling PlatformIO, but the issue persists.
- The same setup works on my Intel iMac.

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-09-01:
it does work since i am developing on a Mac M3. Close VSC delete hidden folder `.platformio` and start VSC

Jason2866, 2025-09-04:
@biologist79 Assuming MacOS, google for installing `brew` install Python with brew. When done delete the hidden folder `.platformio` Now the Python installed from brew will be used for setting up Platformio

biologist79, 2025-09-04:
@Jason2866 Thanks for your answer!
Yes, it's Mac OS. Sorry, forgot to mention.

Is brew necessary? Just asking because Macports is already running with having python 3.13.7 in place (/opt/local/bin/python3).

Jason2866, 2025-09-04:
No MacPorts is fine. Make sure the MacPorts Python is in Path so that the Platformio Installer does find.

Jason2866, 2025-09-06:
Updated pioarduino VSC extension to v1.1.0.  The plugin installs now Python from https://github.com/astral-sh/python-build-standalone when `use inbuilt Python` in settings is checked (default).

Migsi, 2025-09-21:
I've just encountered this issue on an amd64 Linux host. The install fails with
```
error: /home/myname/.platformio/packages/tool-esptoolpy does not appear to be a Python project, as neither `pyproject.toml` nor `setup.py` are present in the directory
Error: Failed to install esptool from /home/myname/.platformio/packages/tool-esptoolpy (exit 2)
```
I tried to understand what is going wrong, even installing the develop version of the platform, but nothing helped.