pioarduino/platform-espressif32 #283: random errors in case of minimal changes

stef-ladefense, 2025-09-06:
I have an incomprehensible problem.
I have a project that compiles without any problems until:
I change the upload port in platformio.ini
or I change a file in one of my libraries (in lib)

I get the following errors.
I have to delete .platformio in my user folder.

my platformio.ini:

```
[env:stable]
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = esp32-s3-devkitc1-n16r8
framework = arduino

monitor_speed = 115200
monitor_filters =	direct
					esp32_exception_decoder

board_build.mcu = esp32s3
board_build.f_cpu = 240000000L
board_upload.flash_size = 16MB
board_build.arduino.memory_type = qio_qspi
board_build.f_flash = 80000000L
board_build.flash_mode = qio
board_build.partitions = partitions.csv
board_build.filesystem = littlefs

upload_speed = 921600

upload_protocol = espota
upload_port = 192.168.10.100

lib_compat_mode = strict
lib_ldf_mode = chain

build_flags =	-O2
				-D CONFIG_LOG_COLORS=1
				-D CONFIG_ARDUHAL_LOG_COLORS=1
				-D CONFIG_TOUCH_SUPPRESS_DEPRECATE_WARN
				-D ARDUINO_USB_CDC_ON_BOOT=0
				-D CORE_DEBUG_LEVEL=ARDUHAL_LOG_LEVEL_WARN
				;-Wno-unused-parameter
				;-fpermissive
				-D CONFIG_ASYNC_TCP_MAX_ACK_TIME=5000
				-D CONFIG_ASYNC_TCP_PRIORITY=10
				-D CONFIG_ASYNC_TCP_QUEUE_SIZE=128
				-D CONFIG_ASYNC_TCP_RUNNING_CORE=1
				-D CONFIG_ASYNC_TCP_STACK_SIZE=4096
	
lib_deps =	esp32async/AsyncTCP@^3.4.7
			esp32async/ESPAsyncWebServer@^3.8.0
			bblanchon/ArduinoJson@^7.4.2

custom_sdkconfig =	CONFIG_COMPILER_OPTIMIZATION_PERF=y
					CONFIG_IDF_EXPERIMENTAL_FEATURES=y
					CONFIG_ARDUINO_LOOP_STACK_SIZE=16384
					CONFIG_LWIP_MAX_ACTIVE_TCP=1024
					CONFIG_LWIP_MAX_LISTENING_TCP=1024
					CONFIG_LWIP_TCP_HIGH_SPEED_RETRANSMISSION=y
					CONFIG_LWIP_TCP_MAXRTX=12
					CONFIG_LWIP_TCP_SYNMAXRTX=6
					CONFIG_LWIP_TCP_MSS=1436
					CONFIG_LWIP_TCP_TMR_INTERVAL=200
					CONFIG_LWIP_TCP_MSL=60000
					CONFIG_LWIP_TCP_FIN_WAIT_TIMEOUT=20000
					CONFIG_LWIP_TCP_SND_BUF_DEFAULT=8192
					CONFIG_LWIP_TCP_WND_DEFAULT=32768
					CONFIG_LWIP_TCP_RECVMBOX_SIZE=32
[... 69 lines cut ...]
C:/Users/Stephane/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libhal.a(gdma_hal_ahb_v1.c.o): in function `gdma_ll_tx_restart':
C:\Users\Stephane\.platformio\packages\framework-espidf\components\hal\esp32s3\include/hal/gdma_ll.h:569:(.text.gdma_ahb_hal_append+0x2e): dangerous relocation: l32r: literal placed after use: (.literal.gdma_ahb_hal_append+0x4)
C:/Users/Stephane/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libhal.a(gdma_hal_ahb_v1.c.o): in function `gdma_ll_rx_enable_owner_check':     
C:\Users\Stephane\.platformio\packages\framework-espidf\components\hal\esp32s3\include/hal/gdma_ll.h:153:(.text.gdma_ahb_hal_set_strategy+0x1e): dangerous relocation: l32r: literal placed after use: .literal.gdma_ahb_hal_set_strategy
C:/Users/Stephane/.platformio/packages/framework-arduinoespressif32-libs/esp32s3/lib\libhal.a(gdma_hal_ahb_v1.c.o): in function `gdma_ll_tx_enable_owner_check':
C:\Users\Stephane\.platformio\packages\framework-espidf\components\hal\esp32s3\include/hal/gdma_ll.h:421:(.text.gdma_ahb_hal_set_strategy+0x3b): dangerous relocation: l32r: literal placed after use: .literal.gdma_ahb_hal_set_strategy

### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

```

Jason2866, 2025-09-06:
OS Windows? If yes it is the path (command) line length limit which results in all kind of different weird issues.

For bigger projects (everything a bit beyond Blink) my recommendation is to use WSL or a VM with Linux with Windows

Jason2866, 2025-09-13:
Closing since no feedback