pioarduino/platform-espressif32 #288: m5stack-core-esp32 renamed in espressif/arduino-esp32 3.x stream

BlueAndi, 2025-09-13:
### PROBLEM DESCRIPTION
In the espressif/arduino-esp32 repository the variant m5stack_core_esp32 was renamed to m5stack_core.
See https://github.com/espressif/arduino-esp32/commit/a1c86ae936c06fd1d7b193986152464cdd4db2ad#diff-498a5473e69c1931584e14350d29bc3f8b22f7cf7469d3ef81c896298c9f9291

Unfortunately the corresponding m5stack-* board json files,  with the variant m5stack-core-esp32 were not updated. Therefore for all of them the compilation is broken because of missing pins_arduino.h

### TO REPRODUCE
Use e.g. m5stack-core-esp32 as board.

### EXPECTED BEHAVIOUR
Compilation is not broken.

### SCREENSHOTS
N/A

### ADDITIONAL CONTEXT
N/A

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-09-13:
As written in readme Boards problems are no issues.

BlueAndi, 2025-09-13:
Oh ... somehow I missed that. Lets keep the issue open, a PR will come.