pioarduino/platform-espressif32 #289: stable pioarduino fails to compile components since idf was bumped to 5.5.1

jnischler, 2025-09-13:
First of all thanks for this awesome work!

### PROBLEM DESCRIPTION
Since the release of 5.5.1 some idf components are no longer working with 5.5.0.
For example:   
espressif/esp_wifi_remote
-> this depends on wpa_supplicant and supplicant was changed
--> https://github.com/espressif/esp-idf/commit/95e3ea8f50a70b8cdd318c10603ad9f4939acf64
--> https://github.com/espressif/esp-wifi-remote/blob/main/components/esp_wifi_remote/idf_v5.5/include/esp_eap_client_remote_api.h#L37

and even older versions have been changed to support the new 5.5.1 types:
https://github.com/espressif/esp-wifi-remote/blob/wifi_remote-v0.16.0/components/esp_wifi_remote/idf_v5.5/include/esp_eap_client_remote_api.h

This makes it impossible to use 5.5.0 together with some components.
And as pioarduino uses 5.5.0 it makes it also impossible to use pioarduino currently.

The develop branch is already using 5.5.1 but we are unable to build our project using develop.
pio run -v fails with:
Error: Failed to install Python dependencies (exit code: 2)
Error: Failed to install Python dependencies into penv

-> could you give me an hint why installing deps is failing with the develop branch ?

### TO REPRODUCE
- create a project with the current stable version 5.5.0
- create file: "# src/idf_component.yml" and add: espressif/esp_wifi_remote: "^1.0.0" as a dependencies
- build

### EXPECTED BEHAVIOUR
I don't understand why espressif made such a "massive" change with just a minor version bump ...
We should bump to 5.5.1 asap.

### SCREENSHOTS
[image]

### ADDITIONAL CONTEXT

**(Please, remember to close the issue when the problem has been addressed)**

Thank you very much.

jnischler, 2025-09-13:
maybe a coincidence but i also needed to add this flag, else the build would also fail:

build_flags =
	-D MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"

Jason2866, 2025-09-13:
This should make no issue. Stable uses fixed Arduino and IDF. Which are matching and known working to compile Arduino as an component of IDF

jnischler, 2025-09-13:
Hey,

Thanks for the fast reply.

Have a look at this files:

5.5.0 version:
https://github.com/espressif/esp-idf/blob/v5.5/components/wpa_supplicant/esp_supplicant/include/esp_eap_client.h

5.5.1 version:
https://github.com/espressif/esp-idf/blob/v5.5.1/components/wpa_supplicant/esp_supplicant/include/esp_eap_client.h

this code block is simply missing in 5.5.0
`/**
 * @brief Bitmask of supported EAP authentication methods.
 */
typedef enum {
    ESP_EAP_TYPE_NONE    = 0,        /*!< No EAP method defined */
    ESP_EAP_TYPE_TLS     = (1 << 0), /*!< EAP-TLS method */
    ESP_EAP_TYPE_TTLS    = (1 << 1), /*!< EAP-TTLS method */
    ESP_EAP_TYPE_PEAP    = (1 << 2), /*!< EAP-PEAP method */
    ESP_EAP_TYPE_FAST    = (1 << 3),  /*!< EAP-FAST method */
    ESP_EAP_TYPE_ALL     = (ESP_EAP_TYPE_TLS | ESP_EAP_TYPE_TTLS | ESP_EAP_TYPE_PEAP | ESP_EAP_TYPE_FAST), /*!< All supported EAP methods */
} esp_eap_method_t;`

And this causes the problems with the wpa_supplicant module

regards

Jason2866, 2025-09-13:
Where did you pin? You cant easily override the idf_component.yml file from Arduino

jnischler, 2025-09-13:
sry i don't use arduino i am using framework = espidf

my platformio.ini:

```
[env:uhf-controller]
;platform = https://github.com/pioarduino/platform-espressif32.git#develop
platform = https://github.com/pioarduino/platform-espressif32/releases/download/stable/platform-espressif32.zip
board = esp32-p4-evboard
framework = espidf
board_upload.flash_size = 16MB
; Force-enable IDF Component Manager during CMake configure
board_build.cmake_extra_args = -DIDF_COMPONENT_MANAGER=1
; (optional) If you also have local components, keep this
; extra_components_dir = components
build_flags =
	-D MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"
platform_packages =
	framework-espidf @ https://github.com/pioarduino/esp-idf/releases/download/v5.5.1/esp-idf-v5.5.1.zip
```

Jason2866, 2025-09-13:
That makes no sense at all. Horrible mixing up stuff. Dont do. When wanting to use IDF 5.5.1 using pioarduino branch develop is working. It uses IDF 5.5.1!

jnischler, 2025-09-13:
i know :) and yes this is horrible :)
But using the develop branch i am just getting this error:

```
Error: Failed to install Python dependencies (exit code: 2)
Error: Failed to install Python dependencies into penv
```

Jason2866, 2025-09-13:
Delete folder `.platformio` and it will clean install.

jnischler, 2025-09-13:
okay thanks will give it a try.

I also noticed that versions up to wifi_remote-v0.15.1 should work with 5.5.0.
Sorry I only looked at the last 5 versions or so :)

jnischler, 2025-09-13:
And one more thing about this line:

```
build_flags =
	-D MBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"
```

Removing this line causes this error:

```
*** Two environments with different actions were specified for the same target: C:\get\repos\get.hardware.uhfcontroller\uhfController-pio\.pio\build\uhf-controller\managed_components\espressif__esp_wifi_remote\dummy_src.c.o                                                                                     
(action 1: riscv32-esp-elf-gcc -o dummy_src.c.o -c -std=gnu17 -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fno-shrink-wrap -fmacro-prefix-map=C:/get/repos/get.hardware.uhfcontroller/uhfController-pio=. -fmacro-prefix-map=C:/Users/j.nischler/.platformio/packages/framework-espidf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -Wno-old-style-declaration -DPLATFORMIO=60118 -DBOARD_HAS_PSRAM -DESP_PLATFORM -DIDF_VER=\"5.5.1\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS dummy_src.c)                                                                                                                                                                                                                                                                                
(action 2: riscv32-esp-elf-gcc -o dummy_src.c.o -c -std=gnu17 -march=rv32imafc_zicsr_zifencei_xesppie -mabi=ilp32f -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Og -fno-shrink-wrap -fmacro-prefix-map=C:/get/repos/get.hardware.uhfcontroller/uhfController-pio=. -fmacro-prefix-map=C:/Users/j.nischler/.platformio/packages/framework-espidf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -Wno-old-style-declaration -DPLATFORMIO=60118 -DBOARD_HAS_PSRAM -DESP_PLATFORM -DIDF_VER=\"5.5.1\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS dummy_src.c)     
```

Adding this line causes only an warning but allows to build and link the project:

```
warning: Two different environments were specified for target C:\get\repos\get.hardware.uhfcontroller\uhfController-pio\.pio\build\uhf-controller\managed_components\espressif__esp_wifi_remote\dummy_src.c.o,
	but they appear to have the same action: ${TEMPFILE('$CC -o $TARGET -c $CFLAGS $CCFLAGS $_CCCOMCOM $SOURCES','$CCCOMSTR')}
File "C:\Users\j.nischler\.platformio\platforms\espressif32\builder\frameworks\espidf.py", line 1030, in compile_source_files

```

Jason2866, 2025-09-13:
Which Python version is installed. Windows can handle correctly only **one** Version installed!
Dont use the Python version from MS Store. It is a crippled (broken) one.

jnischler, 2025-09-13:
yea .... only solidworks and allium is holding me on windows.

I'm really starting to dislike it more and more every day.

Will look into using pio with wsl and remote build ...

But strangely why is the stable version working on windows ... must be a difference.

and why don't i see more logs even with -v ...

very interesting 😜

jnischler, 2025-09-13:
> Which Python version is installed. Windows can handle correctly only **one** Version installed! Dont use the Python version from MS Store. It is a crippled (broken) one.

No did not install using MS Store, just normal python installer.
Python 3.11.6

> Oh my Windows. Install WSL and all your issues will be gone. All tools and toolchains used for to compile and build espressif stuff are coming from Linux. So there are not everywhere workarounds for Windows limitations. The path is probably too long:
> 
> ```
> PS C:\get\repos\get.hardware.uhfcontroller\uhfController-pio
> ```
> 
> Windows has very low limits. Known issue of Windows...

Just for testing i moved the project to:
C:\get\uhfController-pio
-> same result

> Yes, for sure there are differences. It is a newer version ;-) With a lot of changes and added stuff. Just one more include path can be the reason for crossing the path limit for Windows. Anyways all examples do build with Windows CI too. You wont regret using WSL Compile times will be up to 5x times faster.

Is it really recommended to use pio with wsl ?
The IDE is still running on windows and also the board is connected to windows.
One would need to map the usb device to the wsl ...
I do like the idea but I am afraid this is even more challenging :)

Jason2866, 2025-09-13:
idk whats wrong with your Windows setup. See https://github.com/pioarduino/platform-espressif32/actions/runs/17701072167

Jason2866, 2025-09-13:
Can you try following. Deinstall Python3. Remove from Path Sys Variable. Install VSC. Install pioarduino VSC extension. Use VSC "Open Folder" to open an existing pioarduino project. The pioarduino extension should install the needed Platformio penv and Platformio for VSC. After a restart of VSC it should working and load all dependencies from the project.

jnischler, 2025-09-14:
> idk whats wrong with your Windows setup. See https://github.com/pioarduino/platform-espressif32/actions/runs/17701072167
> 
> Maybe the dot in path `C:\Users\j.nischler\.platformio\` In Linux this is a special char. Means hidden. Could be that `j.nischler` makes things go wrong.

very strange, my windows setups is quite clean but you never know :)

And the "." inside j.nischler shouldn't be a problem also not on linux as only a starting "." is causing a hidden folder, not a "." inside a folder name.

> Can you try following. Deinstall Python3. Remove from Path Sys Variable. Delete evey Platformio folder. Install VSC. Install pioarduino VSC extension. Use VSC "Open Folder" to open an existing pioarduino project. The pioarduino extension should install the needed Platformio penv and Platformio for VSC. After a restart of VSC it should working and load all dependencies from the project. It looks like the Python on the System can not generate a working virtual python env (penv)
> 
> ```
> Error: Failed to install Python dependencies (exit code: 2)
> Error: Failed to install Python dependencies into penv
> ```

Okay i am currently cleaning everything and doing a clean install of the extension.

jnischler, 2025-09-14:
VSC output:

```
Installing PlatformIO IDE...
It may take a few minutes depending on your connection speed
Please do not close this window and do not open other folders until this process is completed.

Debugging information is available via VSCode > Help > Toggle Developer Tools > Console.
PlatformIO IDE installed successfully.

Please restart VSCode.
```

After restart, project is loading and installing all dependencies:

```
Resolving uhf-controller dependencies...
Platform Manager: Installing git+https://github.com/pioarduino/platform-espressif32.git#develop
git version 2.45.1.windows.1
Cloning into 'C:\Users\j.nischler\.platformio\.cache\tmp\pkg-installing-l1w7mr4f'...

Platform Manager: espressif32@55.3.30+develop.sha.39f9a30 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp_install/releases/download/v5.3.2/esp_install-v5.3.2.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esp_install@5.3.2 has been installed!
Tool Manager: Installing https://github.com/pioarduino/esp-idf/releases/download/v5.5.1/esp-idf-v5.5.1.tar.xz
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: framework-espidf@3.50501.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/riscv32-esp-elf-14.2.0_20241119.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: toolchain-riscv32-esp@14.2.0+20241119 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esptoolpy-v5.0.2.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: tool-esptoolpy@5.0.2 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/mklittlefs-3.2.0-new.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: tool-mklittlefs@3.2.0 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/cmake-4.0.3.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: tool-cmake@4.0.3 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/esp-rom-elfs-20241011.zip
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-esp-rom-elfs@2024.10.11 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/ninja-1.13.1.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: tool-ninja@1.13.1 has been installed!
Tool Manager: Installing https://github.com/pioarduino/registry/releases/download/0.0.1/scons-4.8.1.zip
Downloading 0% 10%
Unpacking 0% 10% 20%
Tool Manager: tool-scons@4.40801.0 has been installed!
Tool Manager: Installing platformio/tool-scons @ ~4.40801.0
Downloading 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Unpacking 0% 10% 20% 30% 40% 50% 60% 70% 80% 90% 100%
Tool Manager: tool-scons@4.40801.0 has been installed!
Updating metadata for the vscode IDE...
Project has been successfully updated!
```

Did take ages at the "Updating metadata for the vscode IDE..." step.
But this looks quite good.

Did a clean:
[... 36 lines cut ...]
Creating ESP32P4 image...
Merged 3 ELF sections.
Successfully created ESP32P4 image.
```

Info output:

```
pio system info
--------------------------  -----------------------------------------------------------
PlatformIO Core             6.1.18
Python                      3.11.7-final.0
System Type                 windows_amd64
Platform                    Windows-10
File System Encoding        utf-8
Locale Encoding             cp1252
PlatformIO Core Directory   C:\Users\j.nischler\.platformio
PlatformIO Core Executable  C:\Users\j.nischler\.platformio\penv\Scripts\platformio.exe
Python Executable           C:\Users\j.nischler\.platformio\penv\Scripts\python.exe
Global Libraries            0
Development Platforms       1
Tools & Toolchains          10
```

I will now try to reinstall python as i also need a global python runtime.

closed 2025-09-14 by Jason2866

boholder, 2025-09-27:
I also encountered this problem (when running `pio project metadata`)

> Error: Failed to install Python dependencies (exit code: 2)
> Error: Failed to install Python dependencies into penv

and I happened to know some python, here is how I **manually** resolve it (tried to find and fix it on source code level (then send a pr), failed, see below):

By searching GH I find the source code of this error message:
https://github.com/pioarduino/platform-espressif32/blob/85062ff9e38468779f63d59ac091f39844dcccb5/builder/penv_setup.py#L327

A good news is, python is an interpreted language that allows you modify library's source code on fly, so I add some debug code in given source, in my local `C:\Users\18253\.platformio\platforms\espressif32\builder\penv_setup.py`:

https://github.com/pioarduino/platform-espressif32/blob/85062ff9e38468779f63d59ac091f39844dcccb5/builder/penv_setup.py#L312-L328

```python
        cmd = [
            penv_uv_executable, "pip", "install",
            f"--python={python_exe}",
            "--upgrade", "-v"
        ] + packages_list
        print(f"cmd: {cmd}")
        try:
            subprocess.check_output( cmd, timeout=120 )
        except subprocess.CalledProcessError as e:
            print(f"e: {e}")
            print(f"e.output: {e.output}")
            print(f"e.stdout: {e.stdout}")
            print(f"e.stderr: {e.stderr}")
            print(f"Error: Failed to install Python dependencies (exit code: {e.returncode})")
            return False
```

Rerun `pio project metadata`, output of my debug code:

```
cmd: ['C:\\Users\\{user}\\.platformio\\penv\\Scripts\\uv.exe', 'pip', 'install', '--python=C:\\Users\\{user}\\.platformio\\penv\\Scripts\\python.exe', '--upgrade', 'https://github.com/pioarduino/platformio-core/archive/refs/tags/v6.1.18.zip']
...tons of uv debug log
  Caused by: Failed to persist temporary file to C:\Users\18253\.platformio\penv\Lib\site-packages\..\..\Scripts\pio.exe: failed to persist temporary file: Access Denied (os error 5)
e: Command '['C:\\Users\\18253\\.platformio\\penv\\Scripts\\uv.exe', 'pip', 'install', '--python=C:\\Users\\18253\\.platformio\\penv\\Scripts\\python.exe', '--upgrade', '-v', 'https://github.com/pioarduino/platformio-core/archive/refs/tags/v6.1.18.zip']' returned non-zero exit status 2.
e.output: b''
e.stdout: b''
e.stderr: None
```

Aha! This is because Windows will lock a file while some processes using it, that lock will deny operations that modify or delete this file (while in Linux we can `rm -rf /` without OS warning). While running `pio project metadata`, Windows put a lock on `pio.exe`, so the command (it wants to replace `platformio/platformio-core` with `pioarduino/platformio-core`)

```
uv pip install --python="C:\\Users\\18253\\.platformio\\penv\\Scripts\\python.exe" --upgrade https://github.com/pioarduino/platformio-core/archive/refs/tags/v6.1.18.zip
``` 
will fail due to that file lock.

There must be some walk-around to resolve this common problem, as many Windows applications can do self-update, but I'm not familiar with it, so let's leave it to you :P  (maybe they all have a standalone updater program for this purpose)

Manually resolve this problem is quite easy, just manually run the `uv` command in terminal, for me it is:

```
C:\\Users\\18253\\.platformio\\penv\\Scripts\\uv.exe pip install --python="C:\\Users\\18253\\.platformio\\penv\\Scripts\\python.exe" --upgrade https://github.com/pioarduino/platformio-core/archive/refs/tags/v6.1.18.zip
```

Jason2866, 2025-09-27:
For all Windows users "landing" here. Recommended install way. Only install Git. Nothing else! After Git Installation is done. Install VSC. Install pioarduino VSC extension. Wait. Restart VSC when told. When finished open an existing Platformio project. pioarduino is ready to rumble. (The needed Python and all deps are correctly set when following this install flow)

boholder, 2025-09-27:
> Windows is a mess at all.

Well, WSL2 isn't a out-of-box solution and finally unusable (due to instability of USB/IP you can't upload image to device). See my struggles:
https://github.com/boholder/esp32s3_play_ground?tab=readme-ov-file#switch-to-wsl2
If you have solution for it, please share, many many thanks <3 <3 <3

TD-er, 2025-09-27:
One of the latest updates of Windows did break the upload/USB functionality with WSL2.
Maybe it has already been fixed, but since I didn't have time to look into it, I reverted that Windows update and blocked updates for a few weeks.

However I have been using USB while developing ESPEasy via WSL2 on Windows 11 for quite a long time already.
It really is a time saver, literally as building on WSL2 is (on my system) > 5x faster and I don't have those issues with path lengths and much less issues with linking due to command line argument length. (though I had that issue  a few weeks ago on Linux, so I may need to look into that again... as with larger projects you can also run into max. command line length on Linux)